VoronDesign/Voron-Trident #3: Corrections for manual

koonweee, 2021-08-16:
This issue can be used to track all neccessary corrections to the manual.
Please add any corrections as a new comment :-)
[image]
^ contributed by Vapour3D Samuel#5741

mhoush, 2021-08-16:
On page 16 there's a label that says "PEPARE 8 EXTRUSIONS" rather than "prepare".

Yeriwyn, 2021-08-16:
Page 44, is there any way to highlight the m3, because with the m5 everywhere it’s easy to miss, and 3 and 5 do look similar

DunarDolorgiet, 2021-08-16:
Feel free to add annotation to https://assets.adobe.com/id/urn:aaid:sc:US:0f574b90-97b2-4771-a208-91fdc6357ab5?view=published

mhoush, 2021-08-16:
Transferred comments over there, cheers!

PR #13 (2021-08-30): Manual Update
Assembly_Manual_Trident
page 132 - corrected extrusion callout
page 195 - changed screw callout to reflect updated model
various pages: spelling

Frame_Upgrade_Trident
page 8 - updated instructions

closes #3
closed 2021-08-30 by KiloQubit