VoronDesign/Voron-Trident #5: Electronics Bay File Naming Convention

Annex-Engineering, 2021-08-16:
Seems as if the electronics bay brackets use a different naming convention to the standard file structure, as when compared to the rest of the repo (2pc vs x2)

This may cause confusion to end users due to how stuff is worded in the manual.

Congrats on your release!
 - Trails

closed 2021-08-16 by raymondh2

FHeilmann, 2021-08-16:
They're named "2pc" because the STL contains both parts of the mount, so the names are, in fact, correct.

-FH

Annex-Engineering, 2021-08-16:
I get that, just stating that it may be worth a comment in the file naming convention if your going to adopt this as a standard to prevent confusion with end users, 

x# = print quantity
#pc = multiple parts in same file, etc

FHeilmann, 2021-08-16:
We'll keep it in mind if it comes up in the future. 

Thanks for bringing it up

-FH