VoronDesign/Voron-Trident #11: X Carriage Frame, Belt Bolts conflicting with extruder asembly.

MadsGadeberg, 2021-08-24:
Hi Voron Team.

It seems that the bolt head is only recessed by 2,2mm.
[image]

[image]
[image]

SHCS M3 bolts require 3mm to be flush installed.
[image]
[image]

The Extruder assembly is the Galileo type but the backplate should be the same on where the issue is.

eddietheengineer, 2021-08-24:
Thanks for letting us know! The Clockwork motor plate has clearances for the screwheads so they don't intersect:

[image]

Galileo STLs generally aren't included with the stock Voron releases, so you will likely have to add a similar feature to the Galileo parts.

MadsGadeberg, 2021-08-24:
Fuck me Eddie. Ty for the update end sorry for the sloppy issue.

A ton of ty for these projects by the way! i respect that!

eddietheengineer, 2021-08-24:
If you can open the CAD--it's always possible that the Galileo CAD with a tweaked motor plate is in the assembly (maybe hidden). I'm not positive but it might be there!

MadsGadeberg, 2021-08-24:
Ill check and leave the result in the comments for others to see.

MadsGadeberg, 2021-08-24:
I used an early version of the Galileo Extruder which didnt have the groves shown on the picture. 
Feel free to close the issue and many ty for help!
[image]

closed 2021-08-24 by MadsGadeberg