VoronDesign/Voron-Trident #12: Proble and nozzle clearance to small

MadsGadeberg, 2021-08-24:
Hi Voron Team.

I have assembled the Trident MGN12 assembly on my 2.4.

The clearance between the Probe and nozzle is causing me issues. I looked in both the Afterburner CAD and the Trident CAD to se the difference. Here is what i found:

The new MGN12 leaves 1mm of clearance:
[image]

The Old Afterburner leaves around 3mm of clearance:
[image]

My build:
The probe is a bit deformed by the heat but the left and back part shows the original shape. I could maby install the probe a tiny bit higher but it does not solve my problem.
[image]
[image]

It seems that the clearance of 1mm is a bit to small to account for assembly tollerances.

eddietheengineer, 2021-08-24:
Yes--this is one of the tradeoffs for the MGN12 axis, you need a different probe as shown here:

[image]

The PL-08 was likely kept in CAD as a reference to show that it wasn't practical to use

stvptrsn, 2021-08-25:
Technically, the PL08 _should_ work if you have a Mosquito hotend.  But it's easier to recommend moving away from the PL series.

MadsGadeberg, 2021-08-25:
I have pushed the PL up as far as i can and it is just enough. I reckon i have below 0,5mm of clearance which is above 0.

closed 2021-08-25 by MadsGadeberg