VoronDesign/Voron-Trident #17: X Carriage Hex Hole Rounded

crashmaxx, 2021-09-28:
[image]

I noticed the bottom hex nut hole has a round cut going into it some depth.

Since all the other carriages have plain hex holes for the nuts, I'm guessing this is an error. It's an easy fix and I don't mind putting in a pull request, but I wanted to check if someone might have intentionally made it that way first.

The same also applies to the V2.4 MGN12 mod.
[https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/arkeet/mgn12](url)

bdbell, 2021-09-28:
The rounded area is to allow for the insertion of a 6x3 magnet to use with the Hall effect end stops.

crashmaxx, 2021-09-29:
> The rounded area is to allow for the insertion of a 6x3 magnet to use with the Hall effect end stops.

I didn't realize that. No problem, I'll close this.

closed 2021-09-29 by crashmaxx