VoronDesign/Voron-Trident #20: Panel DXFs seem to have a units issue.

BerzerkaDurk, 2021-10-03:
When I open the any of the files in Drawings_DXFs/Panels, they load with the units seemingly scaled incorrectly by a factor of 25.4:1.

For example, the Trident 350 back panel opens with a width of 12,166.6 mm. If I then change the units to inches, I get the correct mm values, but with the wrong units. For example, the above becomes 479 inches.

This happens in both SolidWorks and eDrawings Pro. The 2.4 DXFs open just fine.

silence48, 2022-07-11:
You need to change the units to millimeters when you import the file.  There is 25.4 mm in 1 inch. Your software is thinking 1 mm is 1 inch, thus it is scaled 25.4:1

BerzerkaDurk, 2022-07-11:
Nice of you to assume I'm an idiot. I see you neither read my post nor tried it for yourself.

BerzerkaDurk, 2022-07-15:
The panel DXFs open fine in Fusion 360. I'm assuming there is some odd bug when trying to import them into SolidWorks.

Accordingly, I'm going to close this issue.

closed 2022-07-15 by BerzerkaDurk