VoronDesign/Voron-Trident #22: M5 bolts on XY joints are chambered instead of bridged

henrikssn, 2021-10-13:
This is originally for the V2, but it was suggested in the engineering ticket on Discord to submit the issue for Trident since it has the most recent gantry.

Issue: The recess for the M5x40 BHCS bolts in the XY joints are chambered. This causes the bolts to slowly creep for me, which means I continuously need to re-tighten them

Fix: I used the "bridging trick" which resolved the creeping.

I'm happy to contribute my fix if possible.

timmit99, 2021-10-13:
Chambered? Do you mean chamfered?

henrikssn, 2021-10-13:
> Chambered? Do you mean chamfered?

Correct, I fixed the title/description (sorry, I'm not a native speaker)

BerzerkaDurk, 2022-05-22:
I believe this was addressed in the R1 update

stvptrsn, 2022-05-23:
Yep, I must have forgotten to link the issue.  Thanks!

closed 2022-05-23 by stvptrsn

BerzerkaDurk, 2022-05-23:
You betcha. Great job with the update.

james5843, 2022-05-28:
For anyone who doesn't know what the cross bridge method it like I did, [this video](https://www.youtube.com/watch?v=W8FbHTcB05w) outlines it in detail. A cool design technique!