VoronDesign/Voron-Trident #28: The Frame upgrade guide should list extrusion lengths for 300 and 350 frames.

bwattendorf, 2021-11-02:
I have found it hard to cross reference the right lengths to convert a 1.8 frame to a Trident frame for a 300mm printer. I also suggest that the Trident letter frame be added to the Sourcing guide for each extrusion.

BerzerkaDurk, 2021-11-05:
Couple points:
1. there are no standard 350 mm 1.8 frames.
2. a. Only the Z, rear, bed rear, bed front, and rear bottom extrusions are modified.
    b. The Z extrusions are the same regardless of build size.
    c. For the bed front and bottom rear extrusions, center is center regardless of build size.
    d. For the bed rear extrusion, it is cut 138mm from the end regardless of build size.
    e. For the rear extrusion, it is 330mm long regardless of build size.

agentk, 2022-10-18:
I struggled with the lengths of extrusions from a Format 350 kit and used the wrong length for H Extrusion first time. I ended up making myself a reference for the extrusions in a 350x350x250 kit. (I might have messed something up but you get the idea)

- A Extrusion: HFSB5-2020-500-LCP-RCP-AV360
- B Extrusion: HFSB5-2020-470-TPW
- C Extrusion: HFSB5-2020-470-AH235-TPW
- D Extrusion: HFSB5-2020-340
- E Extrusion: HFSB5-2020-430
- F Extrusion: HFSB5-2020-470-AH235
- G Extrusion: HFSB5-2020-332-LTP
- H Extrusion: HFSB5-2020-330-LTP

[image]

BerzerkaDurk, 2022-10-18:
Yeah. You definitely have the A and B extrusions mixed up on both your chart and picture. This issue was specifically raised for the upgrade guide, so you should probably create a separate issue for this, if you want. I was sure that they'd listed the extrusion letters for each size in the sourcing guide, but i must've dreamt it.

Tuxedo-VE, 2023-04-03:
Actually I think agentK has it exactly right. The BOM and the manual both specify 4 B extrusions ( which have 3 sets of screw head access holes but no threading on the ends) and 9 A extrusions (which have no screw access holes but are threaded on both ends) + 1 C extrusion (which are the same as the A extrusions but with a hole in the middle). The B extrusions form the 4 main uprights of the frame while the A and C extrusions form all the fixed horizontal pieces of the frame. Trust me I spent hours unravelling all this so I know.

IMHO the Trident assembly manual is badly in need of a graphic similar to the one he made in his post. It would have saved me and a lot of other first time builders a ton of time.

itsgudenuf, 2023-08-22:
Just wanted to say thanks to @agentk, that image was worth a thousand words. I was struggling to figure out the various extrusions after opening my Fysetc "kit". After finding this issue, it was a snap to identify them all.

wa2mze, 2025-07-23:
I have a v1.8 300mm frame kit.  This is a 250mm high (Z) size.  I will have to cut some of the extra 420mm extrusions from the v1.8 down to size for the 282 and 330 size extrusions, the 1.8 frame has a 380mm already.  The 330mm  H extrusions can be used as a spacer when mounting the side C extrusions, if it is mounted last.  The E extrusion is a different length than the one on the 250 size trident.   The diagram shown above is for the 350 size trident, the extrusions letters are correct.