VoronDesign/Voron-Trident #39: Trident assembly manual is low contrast, hard to read

mapsedge, 2021-12-14:
The text in the Trident assembly manual is gray, and extremely hard to read for those of us with diminishing eyesight. I know gray is fashionable and "cool", but black on white is easier to read, and ADA compliant.

FHeilmann, 2021-12-15:
Thanks for bringing this to our attention, we'll forward this to the documentation team for consideration. In the meantime I want to humbly suggest you make use of the High Contrast mode included in Adobe's PDF readers to help with this issue.

-Florian

ptyork, 2022-01-19:
Came here to make this very report. Amazed to find someone else had already reported it. It truly is hard to read both with a white background. And it doesn't work at all inverted (which is how I need to read it). The "high contrast" modes in Adobe Reader (and FoxIt and Xodo) do nothing to help with the contrast for these documents. Not sure why. The font is also tiny and thin, all things incompatible with low vision.

I know this is far from a common issue. Perhaps you could simply make the source files (docx??) available on GitHub, those of us with these issues may be able to help ourselves by like Ctrl+A and change the color.

tknapstad, 2022-02-24:
I completely agree. This is an assembly manual, low contrast colors have no place here.

Those who are not on Linux like me can use Adobe Reader to change colors. "Preferences -> Accessibility -> Replace Document Colors + Custom Color" will let you set it to white/black or black/white.

I would also like the source document to be published on Github. A .docx or whatever is something I can open and edit under Linux as well, and it's much easier to annotate to add personal notes than a PDF.

tknapstad, 2022-02-25:
> Those who are not on Linux like me can use Adobe Reader to change colors. "Preferences -> Accessibility -> Replace Document Colors + Custom Color" will let you set it to white/black or black/white.

I tried this on a Windows laptop and it didn't work either, so this needs to be fixed at the source. None of the accessibility tools in Adobe Reader, Foxit Reader or Okular are able to modify the font colors of this document.

DunarDolorgiet, 2022-02-25:
Please have a look and see if **https://github.com/VoronDesign/Voron-2/blob/Voron2.4/Manual/Assembly_Manual_2.4r2_highcontrast.pdf** addresses your needs. If it does, I can include this styling in the next update of the document.

As for accessibility options in Adobe Reader, I'm using the following settings which appear to work fine on my end:
[image]

ptyork, 2022-02-26:
> Please have a look and see if **https://github.com/VoronDesign/Voron-2/blob/Voron2.4/Manual/Assembly_Manual_2.4r2_highcontrast.pdf** addresses your needs. If it does, I can include this styling in the next update of the document.

Thanks, yes, that is much, much better.

> As for accessibility options in Adobe Reader, I'm using the following settings which appear to work fine on my end: [image]

Yeah, depends on the PDF viewer, platform and needs. I need inverted text. Adobe Reader doesn't do a very good job with inverting images. And by not good job, I mean you can't see the images in the manual.

I do have better luck with 3rd party readers sometimes. Found one (Xodo) that can work for most pages on the PC. Still looking for a good one on my iPad, though. 

I DO however see the high-contrast (non-inverted) options ARE in fact working for me in Acrobat DC (paid) but were NOT working in Acrobat Reader DC (free). Perhaps this is the reason for the discrepancies in what we are reporting?

Regardless, if you can switch to the high-contrast version above (or offer it as an option) that would be fantastic.

Thanks a lot!!

closed 2022-05-19 by KiloQubit

atoomnetmarc, 2023-09-30:
It would be awesome if the high contrast manual returned. 

I saw it here: https://github.com/VoronDesign/Voron-Trident/tree/b87f1adac669c8e8c0e9c21d428924088ad0a51a/Manual

It was deleted in this commit: https://github.com/VoronDesign/Voron-Trident/commit/79750725c457ceb16260322cf83922eb3b04ed33#diff-767c04284c07637f52f97edcba7e2014f0afc35131652ec6e965673a6595c656