VoronDesign/Voron-Trident #43: Squaring instructions note to move the gantry to A and B drives to square.  B drive has protruding Y endstop bracket.

mobilemcclintic, 2021-12-19:
In the manual on page 118, we are instructed to move the gantry back to the A/B drives to square the gantry.  The gantry hits the Y endstop block before hitting the A/B drives.  This makes the travel on the right side shorter than the left and had me wondering how in the world I could rack the gantry but always have the left side not be the one stopping.

mobilemcclintic, 2021-12-19:
[image]

closed 2022-02-12 by DunarDolorgiet

j-heffron, 2023-08-05:
The change in the assembly manual noting to rack the gantry to the front idlers rather than the AB drive motor assemblies appears to have been reverted by #68. As far as I can tell, there has not been any change to the y endstop bumper part to warrant reverting the instructions.