VoronDesign/Voron-Trident #46: XY End-Stop Pod wiring length

etjensen, 2022-01-09:
The manual p.203 suggests 150mm of wire for the end stop switches. Not. clear what is intended for where this length is supposed to get you. It would be better to leave the XY end stop pod wiring until later when the Y drag chain is in place so the wire length can be measured out (or give actual wire lengths for the standard builds.

raymondh2, 2022-01-09:
This is so you can add a connector to the spec stand alone microswitches so you can disconnect them without unsoldering them

closed 2022-01-09 by raymondh2

etjensen, 2022-01-09:
Cool- makes sense in retrospect. Adding a comment on p.203 that says "Later we will add connectors to these wires" or something would help. I ended up running long wires as it never occurred to me.