VoronDesign/Voron-Trident #49: A list of suggestions for the manual from my Trident build

Sparohok1, 2022-01-17:
I just finished my build. I thought the manual was great, super clear, clearly a ton of work went into it. Still there were a few places I had notes or nitpicks.
p. 10/11/12 - little tables with lengths for each model would be helpful (or formulas e.g. "370 + D" where D is 0, 50, 100 for 250, 300, 350 builds)
p. 18 - 175 + D/2
P. 54 - 175 + D/2
p. 115 - will this procedure really make gantry square without removing hall endstop mount? for me that hits first!
p. 121 - a table of "typical" and "maximum" belt lengths for each frame size would be really useful. (for 300 build: https://discord.com/channels/460117602945990666/551488536256184331/926551021193408524)
p. 121 - mention correct tensioner position for this procedure (e.g. position tensioner in the middle of its range of motion before measuring)
p. 127 - mention cutting holes for 3 screws in the magnet if not already die cut
p.130 - for me this was not a press fit. but heating up the ABS with heat gun made for easy fit
p.132 - Labelling of extrusions is wrong (https://discord.com/channels/460117602945990666/551488536256184331/918558358896775199)
p. 149 - mention which direction to install heater and thermistor (its implied on p152.., see https://discord.com/channels/460117602945990666/551488536256184331/926588851298906163)
p. 150 - chamfer ptfe tube?
p.157 - mention set screw on drive gear
p.199 - mention m4x6 screws for omron?
p.209 - N and L are reversed for RS-25-5

I see that some of these are already filed as issues in the github, sorry about the duplicates but this is the list of what I saw in my build.

Sparohok1, 2022-01-17:
Let me know if there's anything I can clarify, or if you'd prefer individual issues for each item.
Also, in the spirit of an open source project, I'm happy to try fixing these myself but I don't see any source document for the manual, just PDF.

BerzerkaDurk, 2022-01-17:
p 115 - https://github.com/VoronDesign/Voron-Trident/issues/43
p 121 - belt length is in BoM
p 121 - no. install belts with the tensioners at their "min" positions. they aren't going to get shorter
p 132 - https://github.com/VoronDesign/Voron-Trident/issues/15 ( closed: already fixed )
p 199 - the screws for mine came with the DIN rail mount. is that atypical?
p 209 - https://github.com/VoronDesign/Voron-Trident/issues/41

Sparohok1, 2022-01-18:
@BerzerkaDurk Thanks! I edited accordingly.
p.121 The belt length in the BOM is much longer than my build. I think that's a good idea, better to order too much than too little, but it caused me some soul searching that my measurement was so far off. See the linked post/thread. A table of _typical_ values would be useful which doesn't seem to be what the BOM values are based off of. But I should hasten to add, like many of these comments I'm just listing things that confused me, I'm leaving it to more experienced to judge whether these are real problems!
FWIW I measured with tensioner at midpoint and it worked well for me. That way I avoided the temptation to "add some slack" and went with my exact measurement. I don't think these belts stretch much at all. But I don't really have a strong opinion on the actual recommendation, just that its something worth specifying one way or another for first time builders!
p.199 - My DIN mount didn't come with screws in the package, they were included in the kit however (Formbot). But if they normally come with the mount it makes sense why its not in the manual!

Sparohok1, 2022-01-18:
There are 7 M4x6 screws in the BOM. 4 on page 192, 1 on page 198, so I'm pretty sure the last 2 are for the Omron on page 199.

closed 2022-02-12 by DunarDolorgiet