VoronDesign/Voron-Trident #53: Cannot Download the .STL files for A or B Drive Frame Lower - Both files appear to be corrupted.

Rwhite692, 2022-01-27:
When attempting to download the STL files for the A and B drive Frames (Only the Lowers), Clicking "Download" on either of these two files, results in no download initiated, just a display of a raw ASCII text file.
[image]

raymondh2, 2022-01-27:
STLs can be binary or ASCII. Right-click and save as

closed 2022-01-27 by raymondh2

Rwhite692, 2022-01-27:
Thanks Raymond, I tried that, however these two .STLs will not open in Cura...returns an error message (File is either corrupt, or contains no model data). I even fired up F360 to see If I could import them, with no luck (Was thinking I could import as ASCII / Export them as Binary .stl) ....I am not well versed in dealing with ASCII .stl files so I'm kind of at a loss here...

Rwhite692, 2022-01-27:
Hi Raymond, FYI, I figured it out, and I should have occurred to me.... Windows was adding a hidden extra ".txt" extension after the ".stl" extension (creating filename.stl.txt, with the .txt being conveniently invisible). Newer laptop and never thought to check this. Best Regards, -Rob