VoronDesign/Voron-Trident #58: Clarify Hall Effect Magnet Installation Process

SkiBikePrint, 2022-02-22:
It would be helpful to call out the process for magnet installation for the hall effect sensor on pages 50 and 109.  Page 50 states "HALL EFFECT ENDSTOP? If you are using a Hall Effect Endstop board remove the highlighted part and insert a magnet in its place".  This infers that it should be done as part of prep but the Hall Effect endstop setup document on the Voron web site instructs to wait to install the magnets until the sensor can be powered up and correct magnet polarity determined.  I am not the first to install the magnets during the build in advance of being able to power and test the sensor.  If you get it wrong you pretty much need to reprint the endstop  and/or the x_carriage_frame_right parts.  Thanks, Glenn.

Macmagoo, 2022-03-04:
Unfortunately I did the same luckily I only have to print the endstop.

kodekill, 2022-03-13:
Luckily I didn't glue my magnets in early, but I did find the installation for the hall sensor confusing. I don't recall there being anything in the manual that refers to putting a magnet in the x cariage, just the breakout for the y direction.

nwalasek, 2022-03-14:
I also suffered this fate. Another secondary issue I ran into is the placement of the magnets - they don't line up with the point in space the sensor wants to trigger from.

BerzerkaDurk, 2022-05-22:
To clarify, the issue with the Y endstop is now on pg 48 of the latest manual. I don't see the X endstop magnet referred to anywhere, but at least that one can be pushed out through the screw hole on the opposite side, if it needs to be reversed.

PR #73 (2022-08-16): Manual update
page 13 - swapped F and G callouts
page 48 - added link to setup instructions
page 100 - added missing hexnuts
page 137 - added link to setup instructions

closes #71, closes #70, closes #58
closed 2022-08-16 by raymondh2