VoronDesign/Voron-Trident #65: Deck Support Clip STL Labeled wrong

AleXthePunK, 2022-03-11:
For the Trident STLs, the deck_support_3mm_x8 and deck_support_4mm_x8 are backwards.

blite, 2022-03-18:
Are these shown inserted backwards in the manual as well? The manual seemed to match the CAD when I checked this, but I could not get them installed the way shown with deck_support_3mm_x8 and a 3mm panel Mandala Rose Works panel.   If am going to try the model for deck_support_4mm_x8 installed backwards.

AleXthePunK, 2022-03-23:
The manual is right, the labels on the STLs are backwards.

stvptrsn, 2022-04-10:
@blite The actual parts were a last second change so didn't get reflected in the manual.  They work the same though, just having one leg longer makes it easier to twist them in to place and made some room for the zip tie anchor point.

Is the reported issue that the file names are switched? The 3mm labeled part should be taller for the 3mm thick deck panel.  The 4mm part has extra room for a thicker panel.  I've double checked the parts and they are named as I would expect them to be.

stvptrsn, 2022-05-02:
Cleared this up in Discord chat.  The parts are labeled correctly.  All panel related stuff are named for the total panel thickness they go with (panel plus foam when applicable)

closed 2022-05-02 by stvptrsn