VoronDesign/Voron-Trident #82: Mains inlet wiring on p236 is backwards

wrwindsor, 2022-10-06:
If you're looking into the plug end of the mains inlet wiring, ground tang up, "neutral" is on the left and "line" is on the right.

see: https://www.electronics2000.co.uk/pin-out/iec.php (IEC 63020 pin assignment for C13/C14)

Therefore, if you're looking at the back of the filtered inlet (for purposes of running wiring), ground tang up, it is reversed -- the "line" tab is on the right and the "neutral" tab is on the left.

The wire coloring in the drawing on p236 is backwards of this.

rh101, 2022-10-07:
I was about to post about this issue as well. 

FYI, the Voron 2 manual shows it with the correct wiring.

rh101, 2022-10-07:
It's definitely incorrect in the Trident manual.  The Voron 2 manual shows it with the correct wiring.

BerzerkaDurk, 2022-11-10:
This has happened before. Issue #45. It was fixed then, but it looks like at some point between then and now, they flipped the orientation of the power inlet 180 degrees in the manual, but they didn't swap the wire colors.

Previous manual, correct wiring:
[image]

Current manual, incorrect wiring:
[image]

BerzerkaDurk, 2022-11-10:
I'm going to start an engineering ticket on the Discord. This needs to be corrected immediately. It's a fire hazard. If someone is using the single fused inlet, wired backward, and there is a short to ground, it will short the hot wire right to ground with no fuse protection.

BerzerkaDurk, 2022-11-10:
It should be noted that the wiring on page 237 will have to be corrected, as well.

SirOink, 2023-05-11:
still hasn't been fixed, may 11th 2023. I see this as a major issue due to the amount of low quality kits out and about combined with self-sourcing from non-eu/us countries (the bom filtered power inlet is hard to find in some places, let alone dual fused filtered inlets)

luckily I'm pedantic about safety, I read my power inlet's data sheet & googled IEC ordering, else I'd have fell for it too as I'm 
a. dyslexic
&
b. don't like watching 4 hour long (or even 10 minute long) video tutorials. I prefer written ones, so the manual has been my gospel. I'm surely not alone, hopefully the others have backup fuses bc the devs don't seem to care about updating the trident manual/trident seems to be the red headed stepchild 🤷‍♂️

nemgrea, 2024-02-08:
fixed

closed 2024-02-08 by nemgrea