VoronDesign/Voron-Trident #91: Screw used the wire cover on the A Drive is too short.

marcellomaia, 2022-12-26:
The recommended screw to attache the wire cover on the A Drive is too small.  Maybe a M3x10 instead of a M3x8 SHCS would be the right choice (page 261 of the manual).

nemgrea, 2022-12-27:
[image]
8mm screws seem to be of adequate length...

if your foam tape is thicker than 1mm that may be your issue?

marcellomaia, 2022-12-27:
I believe you did not understand the part I'm referring to.  I'm talking about the screw marked below,  the one the secures the wire cover on the A drive. The distance between the base of the screw to the top of the A drive is 7mm,  so an 8mm screw I believe is not appropriate.

[image]

[image]

nemgrea, 2022-12-27:
[image]
page 225 it looks like then

atoomnetmarc, 2023-10-13:
Current revision has the error on page 231. M3x8 did not fit, M3x12 did fit.

[image]

commit cc1d7651 (2025-06-15): updated cable cover to fix #91 (#184)
closed 2025-06-15 by nemgrea