VoronDesign/Voron-Trident #92: a_drive_frame_lower and b_drive_frame_lower STLs are unusually large (>7 MB) comparing to other STLs

pajtaz, 2023-01-31:
As stated in the title, a_drive_frame_lower.stl and b_drive_frame_lower.stl are large comparing to _upper STLs in the same folder. Importing those two STLs in command line Plater, for example, always throws an error. They seem to be "fine" when opening in a slicer.

I recommend to check those files and see if something went wrong during export to STL. Current workaround for me was to export those files as STLs from the CAD.

HyJeremy, 2023-02-23:
Files were uploaded in STL ASCII format, rather than binary. I'm unsure how to convert, so it would be great if this were fixed

nemgrea, 2023-03-01:
fixed
https://github.com/VoronDesign/Voron-Trident/pull/101

closed 2023-03-01 by nemgrea