VoronDesign/Voron-Trident #94: Keystone module openings in the Skirt have the wrong opening size.

rselec, 2023-02-05:
Hi, I just figured out that the keystone module openings in the Skirt (of the official Trident VTr1 release) have the wrong opening size. My keystone modules fall out.

Opening for clip should be 14.9 x 19.5 mm and is 14.9 x 20.65 mm in the released CAD.

The opening for the front shall be 14.9 x 17 mm and is 14.9 x 18.85 mm.

I'll fix this for me and print the skirt part again. Will someone fix this in the next release? I use Design Spark Mechanical and cannot export a STEP file, STL will be added here.

rselec, 2023-02-05:
[keystone_panel_fixed_openings.zip](https://github.com/VoronDesign/Voron-Trident/files/10611529/keystone_panel_fixed_openings.zip)

rselec, 2023-02-05:
The Keystone Blank Insert is also wrong in size, it would be too large for the fixed openings, needs to be decreased by about 1.2 mm in height.

newphreak, 2023-04-22:
Tested, your fix fits like a glove.