VoronDesign/Voron-Trident #95: Extrusion lengths, including for a 250mm frame, are missing in Assembly_Manual_Trident.pdf

tkalbitz, 2023-02-14:
In the current manual of the Trident are the length for the different extrusions missing.  The pages are missing. They were in the manual in the past see this "Steve Builds" stream: https://youtu.be/JWLwHDgkYPw?t=77

The lengths are needed to reference extrusions length and letters e.g on Page 13.  Additional on Page 13 in the manual is also a reference to the page "The length of marked extrusions depends on the build size. The size shown is for a 250 size Trident. Add 50/100mm for a 300/350 sized Trident."

tkalbitz, 2023-02-14:
One possible solution could be to reference on page 13 to the configurator on the web page and add in the configurator after every extrusion a note "Extrusion 'A' in the manual xx mm".

nemgrea, 2023-03-01:
they were replaced/removed intentionally by the current length diagram on page 13 due to the majority of the extrusion lengths being build size dependent.

the exact number measurements are not needed in order to organize the extrusion as it is shown on page 13

closed 2023-03-01 by nemgrea