VoronDesign/Voron-Trident #119: Wiring PE to frame does nothing.

cmdremily, 2023-05-13:
On page 238 of the Trident assembly manual, the instructions show the PE being connected to the frame without further instruction.

While it is correct from an electrical safety standards point of view to connect any electrically conductive casing of a device to PE. In practice, the aluminium extrusions used on the absolute majority of builds are anodized and that makes the surface of the extrusion non-conductive (just like paint is). Don't trust me? Pick up a multi meter and measure two points on the surface of your nearest anodized extrusion. You'll find the surface has really poor electrical conductivity.

What this means is that simply screwing the PE wire to a roll-in t-nut on the frame (which is what I perceive as the intended solution) might not actually connect the frame to PE unless the bolt used scratches of the anodization and makes electrical contact this way. But there is no instruction to this end.

Furthermore, because of the blind joints design, it's likely but not guaranteed that extrusions that are joined make electrical contact via either the screw and the exposed aluminium in the non-threaded through hole or by scratching the anodization of the non-threaded extrusion. Again, subject to if the screws are anodized themselves. In my build, they do make electrical contact with the exception of the back side horizontal member between the A/B motor mounts.

I would like to see this section of the manual improved to mention both the recommended fastening technique (as it's not obvious to many people that the extrusion isn't conductive) and to use a multimeter to check conductivity from various frame elements to the PE connection (e.g. PSU casing).

melufre, 2023-05-13:
hello 
from my recent expérience don't succeed to get a good contact using T-nuts but  using hammerhead and tigth them strong i was below 3 ohm betewn 3 contact points

rbricheno, 2024-05-13:
Slightly abrading the surfaces of the extrusion works great. This should really be done at all points where two extrusions join. I did this and my whole frame is now continuous and easy to earth. At the very least this should be recommended for the wire from the Wago to the frame.