VoronDesign/Voron-Trident #154: Issue with extruder. It freezes up when printer is powered and won't work when activated

HeATThER, 2024-01-27:
I've tried everything. swapping cables, motors, drivers, and ports to troubleshoot the problem, but nothing is working, leaving me thinking that it must be in the config. This is a new build. The extruder worked last week on a small test run, but after I troubleshooted and calibrated some other things, it now freezes when I power on the printer. I'm truly at a loss on this and need some guidance. The only thing I remember doing out of the ordinary is updating Klipper, Moonraker, and such. It comes with a warning that it can make changes to the system, but I wouldn't know exactly what all the updates did. That stuff is a little above my head.

Thanks for taking a look at my code. I really do appreciate the help. My social network for 3D printing is me and google. XD And I've about exhausted my resources on this one.

[printerconfig.txt](https://github.com/VoronDesign/Voron-Trident/files/14073458/printerconfig.txt)

HeATThER, 2024-01-29:
After some fiddling, I found the problem. I needed ! on the en pin.

timmit99, 2024-01-29:
That would do it! Since the example config has it I'm assuming it just got accidentally deleted during your attempts to get the motor working :-)

Glad to hear it's resolved!

closed 2024-01-29 by timmit99