VoronDesign/Voron-Trident #182: current electronics bay design isn't compliant in Germany (and possibly elsewhere)

Nathan22211, 2025-05-30:
While I know this isn't required in the US and by the EU in general. Talking to Blackbird in Discord, who's German, brought up that their electrical codes require high voltage and low voltage electronics to be physically separated, i.e. on different faces of a machine.

I'd implore that an optional backpack mod is made if this is the case in a user's home region.

this also affects the 2.4, I don't think the V0 has anything considered high voltage and I know nothing of the switchwire or legacy to determine if they're affected as well.

nemgrea, 2025-05-30:
can you please cite the relevant code or provide a source that we can look over? this seems like a very simplified explanation and one that would present problems for a typical PSU as that in itself contains both mains and low DC voltage terminals. I find it hard to believe that youd have to bisect the PSU in order to "separate" high and low voltage....

we need more information here...

Nathan22211, 2025-05-30:
found this https://cms.law/en/int/expert-guides/cms-expert-guide-to-electricity/germany and this https://www.baua.de/EN/Topics/Work-design/Safe-products/Electrical-products/Requirements/Conformity 

basic search query if you want to look into it further: https://www.bing.com/search?pglt=299&q=does+germany+require+high+voltage+and+low+voltage+components+be+physically+seperated&cvid=733cbc40769647afb484030564bc9180&gs_lcrp=EgRlZGdlKgYIABBFGDkyBggAEEUYOdIBCTE2MTc1ajBqMagCCLACAQ&FORM=ANNTA1&PC=U531

blackbird is a electrician over there so he could give info as well

nemgrea, 2025-05-30:
these are both rather vague and not really answering the question of what specifically it is that is in violation....

you HAVE to run mains wiring from the AC parts to the PSU and the PSU HAS to have DC and AC terminals next to each other...

where is the violation occurring? what is the threshold for "High Voltage"? where is the "different faces of a machine" claim coming from?

BerzerkaDurk, 2025-06-19:
How are you defining the terms "high voltage" and "low voltage"?

To answer one of your questions, @nemgrea, in Germany, "High Voltage" is defined as voltages over 1,000VAC or 1,500VDC.
"Low Voltage" refers to 50VAC to 1,000VAC or 120VDC to 1,500VDC.
Voltages under 50VAC and 120VDC are referred to as "Extra-Low Voltage"

In any case, unless I'm mistaken, the regulations you are referencing only apply to devices bearing the CE designation.