VoronDesign/Voron-Trident #185: xy joint mislabel

Spinnertebug, 2025-08-02:
The xy joints found under the path(s)

STLs/Gantry/X_Axis/XY_Joints/xy_joint_left_lower_MGN12.stl 
STLs/Gantry/X_Axis/XY_Joints/xy_joint_left_upper_MGN12.stl
STLs/Gantry/X_Axis/XY_Joints/xy_joint_right_lower_MGN12.stl
STLs/Gantry/X_Axis/XY_Joints/xy_joint_right_upper_MGN12.stl

Do not have the hole spacing for MGN12H or MGN12C, The github file tree name on this may be referring to the linear rail on the "x axis" linear rail itself, however this is misleading and may cause confusion among new builders.

shiftingtech, 2025-08-02:
There are no accommodations in the bom, the manual, or even any kit (that I'm aware of) for mgn12 y rails, whereas there are, historically, accommodations for a few different x axis configurations.  The file names are consistent with those things that actually exist within the voron ecosystem

shiftingtech, 2025-08-02:
Since this has been recently discussed in  #181 , I think there's no need for a further rehash.

closed 2025-08-02 by shiftingtech